LongMetal-Robotics/ShowBot #2: Implement communication with an associated Arduino

i-am-gizm0, 2019-10-30:
The normal RSL is alright, but we can do better. I'm currently working on a program for an Arduino to:

1. Flash some LEDs in time with how the normal RSL should (basically a pass-through)
2. Set other LEDs to different colors based on either the status of the RIO or data in the program

i-am-gizm0, 2019-10-30:
Maybe when the robot is in forward/backward driving mode, the status LEDs could be either green/red.

i-am-gizm0, 2019-11-01:
For now, I am putting this in the Testing column in the Project so we test whether this actually works. Once we verify that it does what it's supposed to, we can move it back to In Progress until issue #1 is closed.
(Estimated ETA: Tuesday 11/5/19 because of our meeting schedule)

i-am-gizm0, 2019-11-05:
I2C isn't working. We'll try Serial over USB and, if *that* doesn't work, we'll resort to PWM or plain digital signals

i-am-gizm0, 2019-11-06:
Successfully changed communication to Serial over USB and tested the Arduino part, so if this doesn't work, it's something with the RIO.

i-am-gizm0, 2019-11-07:
This is tested and working. Now, we need to implement the status functions in the appropriate places in the code (`autonomousPeriodic`, `teleopPeriodic`, and when switching between forward and reverse driving)

Sky14r, 2019-11-07:
Attempted to add requested code after ln.s 59 and 65 as per request. If it suddenly is not doing the thing, it's probably that.

i-am-gizm0, 2019-12-30:
Shooting should be working. When we test that and it is confirmed working, we can close the issue

i-am-gizm0, 2019-12-31:
`PROBLEM` should be working. When we test that and it is confirmed working, we can close the issue

i-am-gizm0, 2019-12-31:
This was tested and verified working (55257c4)

closed 2019-12-31 by i-am-gizm0