LongMetal-Robotics/ShowBot #10: Add Preferences to enable/disable certain subsystems

i-am-gizm0, 2019-11-17:
WPILib includes a class called `Preferences`, which allows for certain settings to be stored across reboots. We can use that to only enable certain subsystems when we want to (include code before the physical part is ready).
For example, if the preference is `false`, it doesn't even create the object or we can use it to change values without changing the code for testing.
[Preferences documentation](https://first.wpi.edu/FRC/roborio/beta/docs/java/edu/wpi/first/wpilibj/Preferences.html)

jon-pun, 2019-11-24:
I imported the Preferences class, and then threw preferences for if the Shooter and Manipulator are enabled under robotInit() in Robot.java. 

Now, you should be able to change what booleans get stored in Preferences in View >> Add >> Robot Preferences. These will be stored when the robot gets shut down, and will be saved in the RoboRIO.

Note: Although the booleans for if the Shooter and Manipulator are enabled exist and can be used, I haven't done anything with them. You can (I think?) just put some "if/else" statements on whether to actually run the elements of the code based on these booleans later.

i-am-gizm0, 2019-12-31:
In the WPILib simulation, this works

closed 2019-12-31 by i-am-gizm0